tht/bpnode-bb #2: Allow BP to reset the RFM69 module

tht, 2017-01-10:
Needs an additional pin, maybe some pull-up/down - unclear...

JohnOH, 2017-01-10:
Consider implications of using the STM reset pin. 

Consider implications of detecting STM software reset since the STM Reset pin is pulled somewhere on software reset. I would be nice to have a latching LED if software reset happened. Pressing hardware reset would not set the LED, it would reset the latch.

JohnOH, 2017-01-12:
7.2.2. Manual Reset
A manual reset of the SX1231H is possible even for applications in which VDD cannot be physically disconnected. Pin 6 should be pulled high for a hundred microseconds, and then released. The user should then wait for 5 ms before using the chip.

Note whilst pin 6 is driven high, an over current consumption of up to ten milliamps can be seen on VDD.

JohnOH, 2017-01-17:
RESET on RFM is active high (ridiculous) - I can't tie them together