tht/bpnode-bb #3: Add a single AA boost chip / LDO regulator

JohnOH, 2017-01-10:
BP is a nice size for AA battery operation.
The existing regulator on the BP is frowned upon as being inefficient.

JohnOH, 2017-01-10:
Perhaps a LiPo charger instead, to act as  a UPS for an AC connected device.

tht, 2017-01-11:
The current *extender* is designed to be used on a breadboard. For standalone use there are some different requirements. For example we don't have to be on top of the BP and the size (specially the width) does not matter that much.

If this board works, we should start to collect requirements for another module - optimized for standalone use. AA power, LiPo charger and also easily accessible SPI and I2C ports are perfect candidates. I'll add a *label* to make sure de don't loose these ideas.

tht, 2017-02-09:
I will close this issue for now as this board has a different use-case and the BP is not optimized for low-power (battery) operation.

closed 2017-02-09 by tht