tht/bpnode-bb #5: Add a SPI Flash chip

JohnOH, 2017-01-12:
Similar to HyTiny extender. Other STM32 boards I have use a Winbond W25X16 which is 2M byte.

JohnOH, 2017-01-31:
Perhaps i2c rather than SPI to keep the radio clear.

tht, 2017-02-11:
Have you a recommendation for a commonly used I2C flash chip? My chunk drawer only *offers* some 5V devices…
I'll probably have to live on the back side of the board.

JohnOH, 2017-02-11:
I am not at home this weekend, I guess the 16Mb chip on the Jeelink would be a contender.

tht, 2017-02-11:
That's a Micron M25P16 SPI (0.80€ each). Looks like there aren't any I2C devices available with this capacity. The biggest one I could find is the STM M24M02 with 2Mbit for about 2.80€ each. So we have to decide between a 2Mbit I2C and a 16Mbit SPI. The board area does not allow for both.
Do you have a specific use-case in mind for this flash module? 

Enjoy your day off!

JohnOH, 2017-02-11:
I imagined using it in a similar manner to the eeprom in the ATMega. I realise that a flash driver could be built (in C) but the cycle count is an order of magnitude less. The STM board I offer to you has an SPI eeprom on it, a W25X16. I was however concerned using SPI as my current approach in RFxConsole involves a scan of eeprom with each received pack as I store details of each node on the network. I don't know if SPI competition would be a problem.

JohnOH, 2017-02-11:
I also think that @jcw is on the i2c route for eeprom.

JohnOH, 2017-02-11:
Rereading everything, 2MB is more than enough for my currently perceived *needs*. Also, I thought you considered eeprom to be out of scope for this board.

tht, 2017-02-11:
Thanks for your feedback John. For me the small board size has priority but if there is a way to squeeze one mir IC on there, I'll do it.
[image]
As you can see the board area is not an issue. But now we need access to PB6 und PB7 for I2C. These pins are (of course) where the RF69 module is. I could add a smd solder pad on the bottom side where the RFM module is. I think this should still be possible to solder onto.

tht, 2017-02-11:
This is how it could work.
[image]
The small hole next to the SMD pads is the antenna connector. I'll move this one away when adding the U.FL connector.

jcw, 2017-02-11:
EEPROM mem over I2C is an order of magnitude slower compared to SPI flash. The F103 48-pin chip has 2 SPI buses, although that means more I/O pins needed, and IMO two devices on SPI is not that hard to support (even with interrupts: you can disable the ones from the radio while accessing flash).

FWIW, I want to drop flash again in the next HyTiny Extender revision - plan is to keep all persistent data on the USB host side. I guess it all depends on your use case.

Most flash chips are compatible, in terms of pinout. Would suggest using a _wide_ SIOC-8 footprint (5.3 mm), I think the narrow ones (3.9 mm) are starting to be phased out, definitely in larger sizes.