tht/bpnode-bb #6: Review CTS/DTR pin selection on FTDI interface

JohnOH, 2017-01-12:
If extender was mounted under the BP we have the option to collect the 4 pin group carrying GND/DCLK/DIO/3v3 in order to use DCLK & DIO similarly to HyTiny extender - see below:
https://github.com/jeelabs/embello/issues/35
This approach supports all the programming options via the FTDI port.

Extender on top would mean light pipes or holes to the see the LEDs - possible.

JohnOH, 2017-01-12:
Excessive enthusiasm on my part. The pin-out for a SER+ should be as above but the pin-in for a JNZ or similar needs to use BOOT0 and Reset.

JohnOH, 2017-01-14:
Review is still desirable since the FTDI port appears **able** to support the BMP firmware. The RX/TX and GND look good - not so sure about the DTR/CTS driving pins and their match to the *standard* BMP image. The 5v needs care to make sure it goes via a regulator on the target or is not activativated.

JohnOH, 2017-01-14:
Looks like these are the BMP pins:
TARGETBOARD     JC66_COREBOARD
 GND             GND
 SWDIO           PA4  * 
 SWCLK           PA5  *
 POWER           3.3V from someplace with a common ground. Probably easiest to plug both boards in to USB

tht, 2017-02-09:
There are some good tutorials for converting a BP into a BMP ([hackaday](https://hackaday.io/project/19238-convert-stm32f103-board-to-a-black-magic-probe)).

For the moment this extension board is for BPs used as a programming *target* on a breadboard. IMHO this kinda conflicts with this issue and I don't want to include complicated parts like light pipes only to get the LEDs back.

Do I miss something here? Do you have a specific use-case?

closed 2017-02-09 by tht

JohnOH, 2017-02-09:
np, and thanks for the hackaday link.

JohnOH, 2017-02-11:
I appreciate that this board can operate as a node, subordinate to a Ser+ using its FTDI interface. I guess it can't operate as a Ser+ because of the choices made with regard to DTR and CTS/RTS. 

I may be losing the plot here.

JohnOH, 2017-02-11:
Ser+ uses the SWD pins which are available on the tail end of the BP. Could use the two SWD pins to drive the DTR/RTS pins on a target node. TXD/RXD still derived from the FTDI header.