tht/bpnode-bb #8: Fix offset of rev1 board

tht, 2017-01-30:
The board is off-center – looks like 1.27mm. Should be fixed in rev2.

jcw, 2017-02-07:
There seem to be slightly different variants of the BP out there, see images.

[image]
[image]
[image]
[image]

JohnOH, 2017-02-07:
I think you may have a red pill layout on a blue PCB. I note that the USB connector is surface mounted rather than through the board. See http://www.stm32duino.com/viewtopic.php?t=117

JohnOH, 2017-02-07:
[image]
[image]

I fourteen units bought in two or three at a time from various chinese sources. They all have the the USB shield attached through the PCB rather than surface mount only.

tht, 2017-02-07:
Same here… about 10 boards, bought from two different sellers. All do have the USB socket mounted through the board. So this is the big difference, good to know. Thanks for investigating this issue. I will add a note in the `README.md`.

jcw, 2017-02-07:
Just to understand: my impression is that there is no difference, apart from appearance/dimensions?

jcw, 2017-02-07:
As I mentioned to Thomas in a private chat, I'm going to call mine the "twist and shout" setup - but it needs a lot of twisting and shouting to set up :) - haven't tried it out yet, it'll probably be 100% fine.

tht, 2017-02-07:
Your's is the one with *personality* :)

There are quite a few differences on these two boards. Labels are different, the USB connector is mounted different and the 2x03 header is not at the same place on the board. I think the board itself should be the same dimension when looking at your pictures. According to http://www.stm32duino.com/viewtopic.php?t=117 there is more like different LDOs and resistor/cap values.

JohnOH, 2017-02-08:
Perhaps we could physically remove four outer pins from the Blue Pill 2x03 header retaining only the path to the relevant GPIO pins. We could then drive these pins with 3v/GND derived from elsewhere on the Expander PCB. This would offer the opportunity of having four holes passing through the expander with connections in parallel such that a Red Pill and a Blue pill could both use the expander PCB soldering the appropriate two pins for that particular board.

tht, 2017-02-09:
Thats a great idea John. I'll most likely go this way for rev2 of the board. So it *should* fit on top of all red- and blue pills in red and blue coats.

3.3V and GND are available right next to these pins, so no problem to get these levels from other pins.

Only disadvantage is the loss in rigidity but I don't think this is a big issue. I will also try to Get hold of some long pin headers. Just snip off the top of these long headers where there is no hole in the extender board and connect all the others through.

tht, 2017-02-11:
I still have to fix the reset button's footprint and check if it's possible to add an U.FL connector but the offset *should* be fixed and this one is compatible with *pills* in all different colours.
[image]
I've added some holes which aren't absolutely needed. This should make it easier to mount it on extra long headers pushed through the BP and this adds-on board.

commit 463dc051 (2017-02-11): Almost ready for rev2. fixes #9, fixes #8, fixes #1
closed 2017-02-11 by tht